HKUST-Aerial-Robotics/VINS-Mono #78: numerical unstable in preintegration & Assertion `fabs(jacobian_speedbias_i.maxCoeff()) < 1e8'

Zekegao, 2017-08-15:
Hi, everyone
I have been trying to solve the IMU pre-intergration recently, but not good. There are two problems, the first is “numerical unstable in preintegration”, another is "Assertion `fabs(jacobian_speedbias_i.maxCoeff()) < 1e8' filed". These two problems arise at same time.

to solve the two problems, I record the IMU pre-intergration results and jacobian_speedbias_i in imu_factor.h.

The IMU pre-intergration tmp_P data before jacobian_speedbias_i too large or too small as below.
[image]
The Tmp_V.x and Tmp_V.z, as below
[image]
My IMU not very good, so I think this result is normal.

But, when the initialStructure() completed, the jacobian_speedbias_i is bad. The Assertion data as below.
[image]
So, I found the residual and sqrt_info are all too large.
the sqrt_info
[image]
the residual
[image]
I have no idea about those. Especially, the position x is figured out just once, -33418258.000000, I'm confused.

Because of I run VINS without ROS, so I have not confirm if those problems are program portable problems yet.
Thanks for your advance.

Zekegao, 2017-08-18:
I found the Ceres max_solver_time_in_seconds is small, so I changed it, and it could work.
So, the reason of too large jacobian_speedbias_i is the max_solver_time_in_seconds.
I also found the covariance in midPointIntegration is defined as zero matrix, could it can be identity matrix?

closed 2017-08-29 by Zekegao

jasminezz, 2017-09-15:
@Zekegao I get the same error as yours, but I did not find a best max_solver_time_in_seconds to run, which value have you used?