HKUST-Aerial-Robotics/VINS-Mono #82: Quaternion local parameter plus and plusJacobian.

pangfumin, 2017-08-21:
Hi, @shaojie  @qintony thank you for sharing this code. Greate Work!
I have one question sbout the quaternion local plus and plusjacobian used in this work.
It seems you are using different quaternion convention against OKVIS, and what Furgale, Barfoot ussually use. Your plus is defined as Q_plus = Q_bar*[0.5 \delta 1]^T. I am confused with the plusJacobian defined[ https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/factor/pose_local_parameterization.cpp ](url) . I think the jacobian should be a 4*3 matrix J=0.5*Left(Q_bar)[I3 03]^T, left represent Quaternion left-hand operator.
Is this right ? Hope for your reply!
Thank you again for the good work!

Pang @Beijing

pangfumin, 2017-08-24:
Hi @qintony , I think I got the answer after some formula derivation，this plusjacobian should be the transpose matrix of the quaternion lift jacobian.  You are right!

Pang

closed 2017-08-26 by qintony

extr15, 2017-12-01:
Hi, @pangfumin @qintonguav , I am still puzzled. 
According to the [`PoseLocalParameterization::Plus`](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/factor/pose_local_parameterization.cpp#L3) in `factor/pose_local_parameterization.cpp`, and the [ceres definition of ComputeJacobian](http://ceres-solver.org/nnls_modeling.html#_CPPv2NK5ceres21LocalParameterization15ComputeJacobianEPKdPd), it is jacobian of `Q_plus = Q_bar*[0.5 \delta 1]^T` with respect to `\delta` when `\delta=0`, so I think it should be `J=0.5Left(Q_bar)[I3 03]^T` instead of the [Identity matrix](https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/factor/pose_local_parameterization.cpp#L20).
Could you explain it to me? Thank you so much!

kafeiyin00, 2017-12-01:
@extr15 @pangfumin @qintonguav I am confused too. Would you please explain it to me.

extr15, 2017-12-01:
OK, I think I know the `Identity matrix` is right now, because the jacobian has already been calculated in `Evaluate`.

QMonteCarlo, 2018-07-02:
Hi, @extr15 @pangfumin can you explain how to explain this local parameterization to me? I'm bothered by this problem too.
I use rotation vector \vec{\theta} to parameterize a unit quaternion, 
q(\theta) = (cos(theta/2), sin(theta/2)\vec{\theta}/\theta), 
then I compute the Jacobian  as:  d ( q(\theta) q(\delta\theta) )/ d\delta\theta　as ceres defines, then
I got the Jacobian equas Jr(\theta).
What's the problem ? I'm confused.

evanrobot, 2019-02-20:
> OK, I think I know the `Identity matrix` is right now, because the jacobian has already been calculated in `Evaluate`.

Hello, I know it's been a long time, but how should I derive the formula of this Jacobian (J=0.5Left(Q_bar)[I3 03]^T), any hint? Thanks

andre-nguyen, 2019-10-21:
To the best of my understanding what @extr15 is saying is that setting the jacobian to identity is essentially the hack described here https://github.com/ceres-solver/ceres-solver/issues/387 where ceres will take the actual full jacobian from the cost function. If anyone else can confirm I would love to know.

shaozu, 2019-10-22:
@andre-nguyen Yes, you are right.

pangfumin, 2022-02-10:
> 

As to  (J=0.5Left(Q_bar)[I3 03]^T,  you can refer to <Indirect Kalman Filter for 3D Attitude Estimation>. But pay attention to the convention, vins use hamilton, but in this article, JPL is applied. This may lead some difference, but it is a good reference.

pangfumin, 2022-02-10:
> As to (J=0.5Left(Q_bar)[I3 03]^T, you can refer to <Indirect Kalman Filter for 3D Attitude Estimation>. But pay attention to the convention, vins use hamilton, but in this article, JPL is applied. This may lead some differences, but it is a good reference.

pangfumin, 2022-02-10:
@duyanwei 

As to (J=0.5Left(Q_bar)[I3 03]^T, you can refer to <Indirect Kalman Filter for 3D Attitude Estimation>. But pay attention to the convention, vins use hamilton, but in this article, JPL is applied. This may lead to some differences, but it is a good reference.

pangfumin-nreal, 2023-02-17:
PlusJacobian * LiftJacobian should be a identity matrix.

956646930, 2025-01-10:
https://www.jianzhuhuai.com/post/lift-jacobian/ This maybe a better explanation