HKUST-Aerial-Robotics/VINS-Mono #84: How to test with my own dataset in offline mode

trinamntn08, 2017-08-21:
Hi,
I want to test VINS on my own dataset in mode offline. In fact, i have a folder which contains sequence images ,a .csv file containing timestamps for each image, another .csv file containing timestamps, gyroscope, accelerometers. Do you know how to put them as input for VINS-system?
I have been blocked in this step for a long time.
I really appreciate your help!

PeiliangLi, 2017-08-21:
Hi, please check https://github.com/PeiliangLi/ios2ros for reference. This repo can read image list and imu data in a file, then publish them as ros topic. The only thing you need to do is record these topics into a rosbag.

closed 2017-08-23 by trinamntn08

BananaGorilla98, 2021-02-28:
Hi,
May I know is there any sample / example that I can refer to? The mentioned link is not available any more when I tried to look for it?