HKUST-Aerial-Robotics/VINS-Mono #86: Run in my own dataset

trinamntn08, 2017-08-23:
i create my own dataset, and i run it by modifying in config file. But it seems that it could not put the image topis in VINS-SLAM. Do i do something stupid here ?

[image]
[image]

[image]
[image]

qintony, 2017-08-26:
Does the image topic contain timestamp?

trinamntn08, 2017-08-28:
[image]
here the header/timestamp in image_raw topic ???

closed 2017-09-06 by trinamntn08

Chandrahas1991, 2017-09-08:
Hi @trinamntn08, i'm facing the same issue. How did you solve it ?
Thanks in advance!

trinamntn08, 2017-09-11:
@Chandrahas1991  in my case, i modified something in code of how to acquire dataset. 
In fact, for the timestamp, before, i got on milliseconds, but got in nanoseconds, it's better.
 And, verified well your datas , timestamps of each images, and timestamps of IMU. Frequences of IMU must exceed 3 times than those of images.

Wanghy118, 2022-06-20:
hi, I met the question, when I test other datasets, it is necessary to convert the frame? and what do I need to change, including datasets and code？could you  help me ?

trinamntn08, 2022-06-20:
> hi, I met the question, when I test other datasets, it is necessary to convert the frame? and what do I need to change, including datasets and code？could you help me ?

Sorry, i've not related on this field for a long time so i couldn't help you out here. Hopefully someone else can help you.

Wanghy118, 2022-06-20:
> > hi, I met the question, when I test other datasets, it is necessary to convert the frame? and what do I need to change, including datasets and code？could you help me ?
> 
> Sorry, i've not related on this field for a long time so i couldn't help you out here. Hopefully someone else can help you.
ok, thank you!