HKUST-Aerial-Robotics/VINS-Mono #88: Help understanding rot_diff

TouqeerAhmad, 2017-08-25:
Hi, I have a question regarding rot_diff being calculated in 'Estimator::double2vector() function' -- lines 518-533 in estimator.cpp

The Rs[0] is the rotation of 0th frame in the current sliding window while the origin_R00 that you are computing that is also kind of coming from the 0th frame of current window? or how they are different -- can you please explain.

TouqeerAhmad, 2017-08-25:
Or is origin_R00 corresponds to the 0th frame in previous sliding window ???

qintony, 2017-08-26:
origin_R00 corresponds to the 0th frame before optimization.
Since we don't fix Rs[0], it will float in the nonlinear optimization due to the unobservable direction.
We just rotate it back after nonlinear optimization.

TouqeerAhmad, 2017-08-26:
Thanks for the clarification!

TouqeerAhmad, 2017-08-27:
Hi, can you please explain the code fragment 686 to 710 in estimator.cpp

While I understand at the high level that you are adding features to the ceres loss function. I am bit confused about 'pts_i' and 'pts_j'. I have checked they are 3d homogenous vectors; are these correspondence between two frames. Now, I have seen e.g. when optimization is called for vision frame 39 or so the feature_index goes from 0 --- 159 (i.e. 160 features) and imu_i/imu_j changes from 0 to 7 and -1 to 6 respectively. These imu_i/imu_j correspond to frames in the current sliding window. 

But the pts_i and pts_j are variable in eact feature_index -- also the first 100 or so out of 160 belong to imu_i =0 and imu_j = -1 and the rest belong to the remaining imu_i/imu_j pairs. 

Here is my objective and why I need this information -- I am trying to replace your vision processing with my own SLAM to give me rotation/translation etc. I understand you apply rotat_diff to V/R/T and I am trying to clarify what information would I need in addition to pose to feed it to ceres at line # 707. 

Any help is greatly appreciated!