HKUST-Aerial-Robotics/VINS-Mono #89: Question about estimated_depth

xukuanHIT, 2017-08-27:
Hi , in the function "void FeatureManager::triangulate(Vector3d Ps[], Vector3d tic[], Matrix3d ric[])" in feature_manager.cpp, what is the meaning of 
[image]
and what the estimated_depth stand for?
[image]

yyIsMe, 2017-08-29:
Q1：
[image]

zhouyanxing, 2017-09-01:
其实就是SVD分解取奇异值最小的那个对应的特征向量作为超定方程的最小二乘解,然后将z坐标转换为齐次坐标即可,所以要除以svd_V[3]

xukuanHIT, 2017-09-04:
ok, thank you !

closed 2017-09-04 by xukuanHIT

poaongithub, 2017-10-31:
hi @yyIsMe,  where the formula you posted come from? could you explain it? thanks a lot.

yyIsMe, 2017-10-31:
@poaongithub 
https://www.zhihu.com/question/29885222?sort=created

poaongithub, 2017-11-09:
@yyIsMe, got it, thanks a lot.