HKUST-Aerial-Robotics/VINS-Mono #91: Question about MINI i7 and NVIDIA TX1 in quadrotor

guanglun, 2017-08-30:
In video.I don't understand the next picture：
[image]
Why use both Mini i7 and NVIDIA TX1 at the same time？I think Mini i7 and NVIDIA TX1 which can be used alone in quadrotor.Only one is enough.So Why?

snakehaihai, 2017-09-25:
read paper. it is quite clear nuc for data input and vins mono stuff. tx1 with connect tec one have usb interface issue, so they use it with ethernet interface to process motion depth

qintony, 2017-10-09:
VINS mono only runs on one computer.  MIni i7 or TX1, either is OK.

TX1 is mainly used for dense mapping, the advanced procedure after visual inertial odometry.