HKUST-Aerial-Robotics/VINS-Mono #93: How to convert the true path coordinates of the dataset MH-05-Difficult into the coordinate system obtained by VINS_Mono?

zhouyanxing, 2017-09-01:
Hi, I'm going to compare the results obtained by VINS_Mono with the true path of the MH-05-Difficult like what you have done in your paper - "VINS-Mono: A Robust and Versatile Monocular Visual-Inertial State Estimator", in Fig.13. But the coordinates system of MH-05-Difficult is different to that of VINS, and I have tried to compare them by Matlab, whose result is that the shape of both of them is similar, but the coordinates and the orientation of the map are different.
So I really want to analyze their errors, can you tell me how to convert it or what the R and T are?
I am very grateful for your help!

Mikhail2017, 2017-09-01:
You can upgrade this script https://github.com/raulmur/evaluate_ate_scale for using with EurocMAV dataset. Maybe it's not necessary to adjust scale, only R, t.

905805952, 2018-01-19:
Excuse me, how do you compare the results, is the tool you write?

905805952, 2018-01-19:
@Mikhail2017 
@zhouyanxing 
Excuse me, how do you compare the results, is the tool you write?

Mikhail2017, 2018-01-19:
No it's tool from one of ORB-SLAM authors. It makes georeferencing for calculated positions and next you have to compare absolute positions and angles for each processed frame.

905805952, 2018-01-19:
@Mikhail2017 
Thank you, could you please send me your tool link?