HKUST-Aerial-Robotics/VINS-Mono #98: [vins_estimator-3] process has died

Sasuke62, 2017-09-13:
Hi, when I first build VINS-mono and run it with euroc dataset, it performs well at first but will crash after few seconds.I've checked version of tools: Eigen3.2.10, OpenCV 2.4.9 ,ROS indigo .Here is the error info:
[image]
The trajectory can also shown in rviz normally at first,how can I solve this problem??Thanks very mach.

MerlotQ, 2017-09-14:
It is hard to say why. Maybe it is better off debugging your run. I think this often happens when memory fails.

deepseamaster, 2020-09-20:
> Hi, when I first build VINS-mono and run it with euroc dataset, it performs well at first but will crash after few seconds.I've checked version of tools: Eigen3.2.10, OpenCV 2.4.9 ,ROS indigo .Here [...]

Hi, Dear Sasuke62. Did you solve this problem? I'm having trouble with it. If you solved it I hope to get our advice. Thanks in advance.