HKUST-Aerial-Robotics/VINS-Mono #101: Question about jacobian of visual part

vonzy, 2017-09-16:
In the code, the jacobian for visual part's residual to attitude is defined as
>         if (jacobians[0])
        {
            Eigen::Map<Eigen::Matrix<double, 2, 7, Eigen::RowMajor>> jacobian_pose_i(jacobians[0]);
            
            Eigen::Matrix<double, 3, 6> jaco_i;
            jaco_i.leftCols<3>() = ric.transpose() * Rj.transpose();
            jaco_i.rightCols<3>() = ric.transpose() * Rj.transpose() * Ri * -Utility::skewSymmetric(pts_imu_i);
            
            jacobian_pose_i.leftCols<6>() = reduce * jaco_i;
            jacobian_pose_i.rightCols<1>().setZero();
        }
In mathematics form, it's 
[image]

but, when I try to derive the equation, I get 
[image]
It's different with the equation used in the code, I don't know whether it's correct and where is the mistake if it's wrong. Please tell me, Thanks!

vonzy, 2017-09-17:
@PeiliangLi @qintony

vonzy, 2017-09-21:
All right, I have got the correct derivation. Don't need to reply. :）

closed 2017-09-21 by vonzy

vonzy, 2019-01-23:
VINS uses right perturbation form

evanrobot, 2019-01-30:
> VINS uses right perturbation form

thanks.