HKUST-Aerial-Robotics/VINS-Mono #102: IMU parameters

derekhe89, 2017-09-18:
Hello, I want to know the IMU parameters(acc_n gyr_n acc_w gyr_w) is continus time or discrete time?

vonzy, 2017-09-18:
discrete

csroccv, 2018-09-12:
in the file euroc_config.yaml,note like:
acc_n: 0.2 # accelerometer measurement noise standard deviation. #0.2
gyr_n: 0.02 # gyroscope measurement noise standard deviation. #0.05
acc_w: 0.0002 # accelerometer bias random work noise standard deviation. #0.02
gyr_w: 2.0e-5 # gyroscope bias random work noise standard deviation. #4.0e-5

but I feel they can't calculate by allen variance,by imu_utils (the tool calculate allen variance).What exactly does they mean?How to get them?How to calculate them?@derekhe @vonzy

maximpavliv, 2020-03-07:
> discrete

Are you sure? What is the explanation?