HKUST-Aerial-Robotics/VINS-Mono #105: Calibration IMU & ZED Camera

sandmen, 2017-09-29:
Hi Gent's,
I have VINS-MONO now running on Jetson TX2.
I have connected a MPU9250 via SPI and a Zed Stereolabs-Camera.
MPU is via rtimulib2 prepared as ROS-node, and Stereolabs-zed-Ros-Wrapper is also working.

Vins-Estimator is roughly working !
Now I need to calibrate the camera and IMU.
Hmm, anyone can help me how to calibrate the camera ?

The Zed Camera it self, I have already calibrated ! but not in ROS!

raghavkhanna, 2017-10-02:
You may want to look at https://github.com/ethz-asl/kalibr/wiki/camera-imu-calibration

snakehaihai, 2017-10-06:
we tried this solution. although sometimes drift. but as a simple solution it will work. 
we use the ros zed driver with vector nav on vins mono. The value are directly taken from config / a3 /ex_calib_result.yaml.   put the value in that text file in the cfg. also remember to set flag to 1.

sandmen, 2017-10-06:
I need to calibrate the onboard mpu9250 from Auvidea J130 board.
Alread have the RTIMULIB-Wrapper in ROS, and provide the values per node.
Hopefully I have time the next days....

yueshukun, 2018-08-29:
@sandmen Do you succeed to run vins-mono with zed and mpu9250.Thanks!

sandmen, 2018-08-29:
@yueshukun 
sorry, I got it running, but I didn't get the mpu calibrated and synced with zed.
But to be honest, Oct 2017 was maybe last time when I was working on this topic !