HKUST-Aerial-Robotics/VINS-Mono #106: Running Ar_demo on TX1

snakehaihai, 2017-10-04:


snakehaihai, 2017-10-06:
emm might found out the solution. need to change eigen and cere version. for different ubuntu version are different. there is still some small-scale drift which could be caused by wide-angle camera calibration. will try to solve it

snakehaihai, 2017-10-09:
ok tried vins on ueye 1221 camera( the same which lsd slam was using)  and A3. it works like magic. With 100 feature and no loop closure on a thin ultrabook. it can stay for half hour without drift. however tx1 still have problem. i think its 16.04, will try to lower the version on tx1 and try it again

qintony, 2017-10-09:
Sounds so good. Thanks for your work.

snakehaihai, 2017-10-10:
Can you pose up your parameter and system setting to run vins_mono on tx1?  we tried on 16.04 with a few version of eigen and ceres  doesn`t works well. can you pose up your setting?

gossterrible, 2017-10-28:
@snakehaihai Hi , where you able to get it working on the TX1 with Ubuntu 16.04 ?

Jafzdf, 2019-01-09:
> ok tried vins on ueye 1221 camera( the same which lsd slam was using) and A3. it works like magic. With 100 feature and no loop closure on a thin ultrabook. it can stay for half hour without drift. [...]
Hi,I want to ask how to run ar_demo on EUROC dataset with VINS_MONO？

Jafzdf, 2019-01-09:
Hi,I want to ask how to run ar_demo on EUROC dataset with VINS_MONO?

snakehaihai, 2019-01-09:
@gossterrible yes it can work on TX1, TX2, ubuntu 16.04 and 18.04(need quite few changes).

snakehaihai, 2019-01-09:
@Jafzdf   use the config file on  vinsmono/config/euroc      change the launch file configs with the one in   vinsmono/config/euroc

Jafzdf, 2019-01-09:
Sorry,I try it but still can not follow you.Are you Chinese?Maybe we can communicate with Chinese if you are .Thank you.