HKUST-Aerial-Robotics/VINS-Mono #107: problem roslaunch vins_estimator euroc.launch

xhmtq, 2017-10-10:
tq@tq-VirtualBox:~/catkin_ws/ceres-bin$ source ~/catkin_ws/devel/setup.bash
tq@tq-VirtualBox:~/catkin_ws/ceres-bin$  roslaunch vins_estimator euroc.launch
... logging to /home/tq/.ros/log/b6554b08-ad28-11e7-ab15-08002793df67/roslaunch-tq-VirtualBox-13047.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tq-VirtualBox:42397/

SUMMARY
========

PARAMETERS
 * /feature_tracker/config_file: /home/tq/catkin_w...
 * /feature_tracker/vins_folder: /home/tq/catkin_w...
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /vins_estimator/config_file: /home/tq/catkin_w...
 * /vins_estimator/vins_folder: /home/tq/catkin_w...

NODES
  /
    feature_tracker (feature_tracker/feature_tracker)
    vins_estimator (vins_estimator/vins_estimator)

auto-starting new master
process[master]: started with pid [13059]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b6554b08-ad28-11e7-ab15-08002793df67
process[rosout-1]: started with pid [13072]
started core service [/rosout]
ERROR: cannot launch node of type [feature_tracker/feature_tracker]: can't locate node [feature_tracker] in package [feature_tracker]
ERROR: cannot launch node of type [vins_estimator/vins_estimator]: can't locate node [vins_estimator] in package [vins_estimator]

snakehaihai, 2017-10-10:
catkin make build error. try rospack list see if you can see the 2 node.

xhmtq, 2017-10-10:
i can not see the 2node,what can i do to make the command regular. i am ros newer ,desire to make the VINS-Mono out.

gossterrible, 2017-10-18:
I had a similar issue and it seems it's was due to the opencv version I was using. After changing the version from 3.1 to 2.4.8 my issue was fixed.

- Check your opencv version 
`pkg-config --modversion opencv`
If your version doesn't meet the test environment then i will advice you to change it to 2.4.8 
> (Our testing environment: Ubuntu 14.04, ROS Indigo, OpenCV 2.4.8, Eigen 3.2.0)

### Attention : 
Don't use `apt-get remove` to remove opencv because this might also affect your ROS installation ,instead download and compile opencv 2.4.8 somewhere else on you computer and then use 
`sudo make install` in the directory where you complied opencv 2.4.8 to install the new version of opencv 2.4.8 on your system.Check to make sure you have the right opencv version.

- Next you will have to use 
`catkin_make`
`source ~/catkin_ws/devel/setup.bash`

mircial, 2017-11-14:
[feature_tracker-2] process has died [pid 17654, exit code -6, cmd /home/liu/catkin_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/liu/.ros/log/30d4f70a-c900-11e7-a798-000c2996b259/feature_tracker-2.log].
log file: /home/liu/.ros/log/30d4f70a-c900-11e7-a798-000c2996b259/feature_tracker-2*.log
[vins_estimator-3] process has died [pid 17661, exit code -6, cmd /home/liu/catkin_ws/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/liu/.ros/log/30d4f70a-c900-11e7-a798-000c2996b259/vins_estimator-3.log].
log file: /home/liu/.ros/log/30d4f70a-c900-11e7-a798-000c2996b259/vins_estimator-3*.log

mircial, 2017-11-14:
How to deal with it?

jn-c, 2018-07-12:
@mircial  I have the same problem as you. Did you fix it?

qimao7213, 2021-01-12:
> @mircial I have the same problem as yours. Did you fix it?

did you fix it?