HKUST-Aerial-Robotics/VINS-Mono #108: Division by 100 in the imu initialization

K0stIa, 2017-10-14:
Could you explain what does this magic 100 number stand for? here https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/initial/initial_aligment.cpp#L151 ?
And later you multiply by 1000. 

Would be interesting to get some explanation behind this. 
Thanks in advance.

vonzy, 2017-10-23:
I think it's to make the scale (usually around 0.01) solved more accurately in the equations

huangju91, 2019-03-29:
> Could you explain what does this magic 100 number stand for? here https://github.com/HKUST-Aerial-Robotics/VINS-Mono/blob/master/vins_estimator/src/initial/initial_aligment.cpp#L151 ?
> And later you multiply by 1000.
> 
> Would be interesting to get some explanation behind this.
> Thanks in advance.

Hi，the word “magic” is just perfect！！！  I tried to remove both“100” and got very horrible path,  do you get the reason, share with me if you do

shaozu, 2019-03-30:
Hi, that is for avoiding numerical issues.

closed 2019-03-30 by shaozu