HKUST-Aerial-Robotics/VINS-Mono #112: how can i use duo3d mlx in vins-mono?

yxyxwanwan, 2017-10-25:
i am confused that how can i use duo3d-mlx in vins-mono ? in duo3d offcial web ,they just provide me with a rosdriver bag,the roslaunch file offers the rectified image topic and imu topic .i think that this rectified image affects the FOV.How can i use the original image topic?

taoyang1994, 2017-11-03:
Can you run VINS with DUO 3D mlx? Is your problem solved?

pxlong, 2017-11-28:
maybe you can try to set distortion_parameters (in config.yaml) to 0.0.

yxyxwanwan, 2017-12-06:
@pxlong hello,it seems you have tested vins with duo3d.may i have your contact information ?such as wechat or QQ? thank you very much indeed.

yxyxwanwan, 2017-12-06:
@taoyang1994 no...

pxlong, 2017-12-06:
@yxyxwanwan  sorry, I do not run vins with any duo 3d camera (I don't have any duo camera). I may can try methods I mentioned in other issues #124 and #116

taoyang1994, 2017-12-07:
你好，我也在学习VINS，想和你交流一下DUO MLX，我的QQ号：1792894646。你能留一下，你的联系方式吗？谢谢。