HKUST-Aerial-Robotics/VINS-Mono #113: problem about std::bad_alloc

JMZZZ, 2017-10-28:
Hello,I want to train my own bag of words for this program with DBOW2, I refer to your brief pattern and the data format in VocabularyBinary.hpp. But When I load my own bag, there is a problem ： terminate called after throwing an instance of 'std::bad_alloc'.

I do not know why this problem occurs, Could you please tell me how did you generate the dictionary?

[image]

You-zeyi, 2020-09-24:
I was wondering if you haved solved this problem. i came cross the same problem when using a 2017 old-version vins-mono

slyrasa, 2021-10-31:
the problem has been solved？please tell me some details