HKUST-Aerial-Robotics/VINS-Mono #114: Question about eigenvalues square roots at the end of Marginalization

rachillesf, 2017-10-30:
Hello,
I have a question about why after the computation of Schur complement the jacobian is computed using the square roots of the resultant A matrix. I'm not able to find the derivation anywhere. 
[image]

Thanks!

rachillesf, 2017-10-31:
Hi, just figure it out.

The operations are meant to extract the Jacobian and residuals from the form : 
A =  J_transpose * J,  and 
b = J_transpose * residual.

Thanks

closed 2017-10-31 by rachillesf

epicjung, 2023-06-18:
Hi, I have a follow-up question about the square-root of Hessian matrix. I found that the A matrix does not necessarily be semi-positive definite matrix, which means that it can have negative eigenvalues as well. From the select function in calculating S, all the negative eigenvalues will become zero. I think this could give a wrong linearized_jacobians and thus cannot accurately change the residual in the marginalization factor (residual_new = old residual + linearized_jacobians * dx). I just want to how to manage this kind of situation. Many thanks in advance.