HKUST-Aerial-Robotics/VINS-Mono #119: how to make euroc format dataset

cumtchenchang, 2017-11-07:
Now euroc dataste is the most famous visual inertial dataset. I want to know how to make euroc format dataset with my own camera and IMU and compare different open source VIO algorthm with this dataset.
                                                                 thank you.

taogashi, 2017-11-13:
Kalibr. No Thanks.

cumtchenchang, 2017-11-13:
@taogashi  你好。我只用过kalibr进行标定摄像头和IMU之间的关系，但是并没有发现能够制作euroc 格式数据集的工具，或者是我没有自信注意。请问你有使用过kalibr制作过吗？效果怎么样，和euroc 的数据集有没有区别。谢谢。

taogashi, 2017-11-15:
kalibr有个[叫bagcreator的工具](https://github.com/ethz-asl/kalibr/wiki/bag-format)，官网上有文档，按照相应的格式做成rosbag就可以play了。

cumtchenchang, 2017-11-30:
非常感谢您提供的帮助，最近一段时间制作了一个双目视觉惯性里程计，根据bagcreator制作了rosbag。现在还是有点疑问，现在很多算法使用euroc的数据包，调用了SEQUENCE.txt文件，我试着制作了一下但是总是会出现错误。不知道您这边能否给予帮助。万分感谢。

duyilong, 2018-04-11:
@cumtchenchang 现在制作数据集成功了吗？