HKUST-Aerial-Robotics/VINS-Mono #124: 我运行的是camera_pose 跳变，这是为什么？（when i running the project ,the camera_pose'data mutation)

Daivesi, 2017-11-21:
excuse me,  I had calibrate the mono camera , but i haven't  calibrate the IMU. because i haven't the IMU parameter . so ....
我用的euroc_config.yaml配置文件，estimate_extrinsic 参数1和2 试过。 camera_pose 还是跳变。
这个相机和IMU 一定要联合标定吗？ 我只标定了相机，IMU的那些需要的.yaml的参数我没有，不知道从哪里得到，还有就是T_C_I 这个IMU装换到camera的矩阵怎么算？

Daivesi, 2017-11-21:
我发现demo里面跑的 bag包 里面imu数据和我用的imu数据有很大出入，哪个角速度Z轴的值，demo里面哪个值变化很小，而我的直线上升，然后变成零又。 同是imu里面的陀螺仪输出的角速度，为什么差别这么大呢？

StevenCui, 2017-11-27:
Try to calibrate the Cam-Imu with Kalibr "https://github.com/ethz-asl/kalibr".

pxlong, 2017-11-28:
I have not calibrated my IMU, but VINS-Mono still works. You may want to try:
1. set your IMU parameters in config file. It depends on the quality of your IMU, for low-end IMUs, the following parameters may work:
acc_n: 0.1        # accelerometer measurement noise standard deviation. #0.2
gyr_n: 0.05         # gyroscope measurement noise standard deviation.     #0.05
acc_w: 0.01        # accelerometer bias random work noise standard deviation.  #0.02
gyr_w: 4.0e-5       # gyroscope bias random work noise standard deviation.     #4.0e-5
g_norm: 9.81007 

2. set estimate_extrinsic to 2

3. launch vins_estimator, and move the camera carefully (rotate it slowly). Once it works, VINS will generate a "ex_calib_result.yaml" (camera-imu extrinsic) file for you. 

4. set estimate_extrinsic to 1, and set extrinsicRotation and extrinsicTranslation using the results in "ex_calib_result.yaml". Walk around with your camera, VINS will update the "ex_calib_result.yaml" . Then, again you can set extrinsicRotation and extrinsicTranslation with the latest "ex_calib_result.yaml" .

Hope this will help

Daivesi, 2017-12-15:
@pxlong 　I think about the IMU parameters  is not big problem . i just wanna  the odometry's data by IMU or by vision ? why i put  and analyse it , which the result unnormal

Daivesi, 2017-12-15:
@StevenCui I think about the IMU parameters is not big problem . i just wanna the odometry's data by IMU or by vision ? why i put and analyse it , which the result unnormal