HKUST-Aerial-Robotics/VINS-Mono #125: IMU parameters configuration

xiaofeifei1234, 2017-11-24:
Hello ,question about imu parameters in the euroc_config.yaml file,comment say that the more accurate parameters provide ,the better performance, so i want to calibration my imu ,but i don't know how to get the matching parameters for acc_n,gyr_n,acc_w,gyr_w, do you use the Allan variance or another methods?  and what's the units of these parameters? Thank you very much!

StevenCui, 2017-11-27:
I have the same question with you. And I am using the Kalibr_allan tool to calibrate the IMU noise "https://github.com/rpng/kalibr_allan#imu-noise-values". But i do not know the result of the Kalibr_allan is suit for VINS.  Look forward to @LiPeiliang's answer.
Another question is when the IMU noise "acc_n acc_w gyr_n gyr_w" of EuRoC MAV Dataset increased or decreased by 10 times, the camera pose of VINS were almost the same. But using my own sensor, the camera pose is sensitive to the IMU noise parameters. Why?

xiaofeifei1234, 2017-11-27:
@StevenCui 
yeah ,i also use the kalibr allan tools but it has a big difference between my calibration results and the euroc.config.yaml,so i am not sure whether my results are correcet or not .As for your question,i think your IMU parameters may be match for the  Euroc MAV Dataset's IMU, if you use your own imu's paremeters the solution may be better.

> 
Kalibr IMU Noise Parameters in Practice

It is important to note that the IMU measurement error model used here is derived from a sensor which does not undergo motion, and at constant temperature. Hence scale factor errors and bias variation caused by temperature changes, for example, are not accounted for. So clearly, the model is optimistic. Particularly when using low-cost MEMS IMUs with Kalibr, you may have to increase the noise model parameters to "capture" these errors as well. In other words, if you use directly the "sigmas" obtained from static sensor data, Kalibr will tend to trust your IMU measurements too much, and its solution will not be optimal.

From our experience, for lowest-cost sensors, increasing the noise model parameters by a factor of 10 or more may be necessary. If you use Kalibr with such a device, please give us feedback, such that we can develop specific guidelines, device-specific parameter suggestions, or more advanced methods to determine these parameters.
>