HKUST-Aerial-Robotics/VINS-Mono #131: Question about the MEI camera model

se7ven7, 2017-12-07:
I found that the camera model provided by you in the config file is MEI.
I wonder if I can use pinhole camera model instead. Or how can I get camera parameters  in MEI format?
Looking forward to your reply~~Thanks in advance~~

xjtao1314, 2017-12-12:
MEI is better than PINHOLE as the papper describes.  and MEI calib parameter can be derived from the project camodocal   https://github.com/hengli/camodocal

duyilong, 2017-12-20:
what is MEI?

qintonguav, 2017-12-24:
In the EUROC dataset, we use PINHOLE model. 
MEI is used in the fisheye camera (large FOV camera), which PINHOLE can not model.

xav12358, 2019-01-09:
Vins Mono seems to support Mei calibration model.

Is there example of dataset and calibration to try it?

Are there also other modifications needed to setup Vins for large FOV cameras?