HKUST-Aerial-Robotics/VINS-Mono #132: PreIntegration corrected  by the change of bias in the paper????Can not understand the Jaccobian .......I think they maybe wrong.

GONGXI1994, 2017-12-12:
file:///home/orbbec/%E5%9B%BE%E7%89%87/2017-12-12%2017:03:37%E5%B1%8F%E5%B9%95%E6%88%AA%E5%9B%BE.png

GONGXI1994, 2017-12-12:
Foster give the full derivation of the Jacobians, however your paper seems different from what he has done, and It would be very kind if you provide the reason why you  can simply get the Jacobian from the F...... @shaojie @dvorak0 @WayneTimer @groundmelon @xuhao1

qintonguav, 2017-12-24:
Calculate F is tough. Jacobian is directly part of F, even in Foster's formulation.

GONGXI1994, 2017-12-24:
I konw the F is constructed by the jaccobians ,but I do not think the "error state "  jaccobian is the same jaccobian used to correct the influence of the change of the bias.....

GONGXI1994, 2017-12-24:
I konw the F is constructed by the jaccobians ,but I do not think the "error state " jaccobian is the same jaccobian that used to correct the preIntegration when the bias change......
@qintonguav

gavinfan1, 2018-05-18:
I also can not understand the Jaccobian here. Can you give more information about the jaccobian ? I can not find the Foster's formulation. Can you give me the network link? @qintonguav  @GONGXI1994

GONGXI1994, 2018-05-24:
paper: On-Manifold Preintegration for Real-Time Visual-Inertial Odometry

struggleforbetter, 2018-12-11:
I'm also puzzled with this problem.The jaccobian should be the measurement change with the bias change.  Why is it this? Are the two Jaccobians same ?

lywcn, 2020-07-23:
I am also confused here. any updates?