HKUST-Aerial-Robotics/VINS-Mono #134: concerns about the FEJ

castiel520, 2017-12-13:
while marginalizing the old state, as the FEJ describe: we need to fix the linearization point.

1. I do see VINS fixes the linearization point around x = x0, n the prior residual part, but in the imu residual part the Jacobian is getting updated around different linearization point(x!=x0), will this cause any consistency problem of the system, or I'm having some misunderstanding on the FEJ? 

Thanks.

pxlong, 2017-12-14:
Hi, what is FEJ?

castiel520, 2017-12-14:
@pxlong First Esitimation Jacobian

vonzy, 2017-12-14:
The x0 you mean should be updated iteratively, any nonlinear optimization will not only linearize the formula for one time.

castiel520, 2017-12-14:
@vonzy what i mean is fix the linearization point around x=x0, I'm not saying we dont update x iteratively.

FEJ indicates that if the linearization point is different for the same variable will bring inconsistecy to the system, say a varible x3 in the sliding window, we have the IMU residul E(imu) and the prior residual E(prior), if we want to do the optimization, we need to linearizeboth E(imu) and E(prior) to the same varible x3, if the linearization point is different then the system will be at stake

vonzy, 2017-12-14:
Get ur point, but I haven't notice that, let's wait!

guofengw, 2017-12-18:
Do you know in the marginalization part, how the linearized_residuals make out? I know linearized_jacobians is make out of matrix A which means linearized_jacobians.transpose()*linearized_jacobians = A.

castiel520, 2017-12-18:
@guofengw b = J*e where e is the linearized_residuals

guofengw, 2017-12-20:
@castiel520 Does you J here is linearized_jacobians? I verify linearized_jacobians*linearized_residuals does not equal to b.

stihuangyuan, 2018-04-18:
@castiel520  https://blog.csdn.net/heyijia0327/article/details/53707261
from this blog, maybe FEJ fixed linearization point just apply prior error.

huangju91, 2019-03-28:
> @castiel520 https://blog.csdn.net/heyijia0327/article/details/53707261
> from this blog, maybe FEJ fixed linearization point just apply prior error.

yes, in my understanding ,both vin and okvis use FEJ only in marginalization error,So I wonder why,and if I wanna use FEJ in all errors,should I fix the linear point only when prior error is included in ceres solver while not fix when no prior is in my solver?