HKUST-Aerial-Robotics/VINS-Mono #139: Map saving/reuse and localisation mode

Changliu52, 2017-12-22:
Hi Guys,

Thanks again for sharing the amazing code.
I wonder if there is any existing implementation for reusing the map and doing localization only in an old map? If not, some suggestions for implementing that would be really appreaciated.

It would be really useful to have this feature. 

Thanks
Chang

qintonguav, 2017-12-24:
Yes, we have this feature. The code will be released by the end of this year.
Thanks!

Changliu52, 2017-12-25:
Thank you Qin for the info.
This is fantastic news for us!
Do you mean the end of 2017 or the end of 2018?

Chang

qintonguav, 2017-12-25:
2017. In this week.

Changliu52, 2017-12-25:
Great! Qin
Please post the link to this thread when it is out, if it is in the different repository.

Chang

closed 2017-12-25 by Changliu52

msamogh, 2019-03-26:
Any updates on this?

Leerw, 2022-08-15:
Hi, any update here?