HKUST-Aerial-Robotics/VINS-Mono #143: numerical unstable in preintegration

wtmilk, 2018-01-05:
In function "img_callback" of program feature_tracker, I publish all feature. “numerical unstable in preintegration” happened to vins-estimator。How to fix it?

BITVoyager, 2018-01-08:
Hi there. The coordinate system of the IMU (body) frame is typically different from that of the camera frame, i.e. there may be a non-identity rigid-body transformation between the two coordinate systems. In particular, the sensor frame definition can be found at [here](https://developer.android.com/guide/topics/sensors/sensors_overview.html#sensors-coords). The camera frame is way tricky because it depends how you acquire and manipulate the image.

wtmilk, 2018-01-08:
@BITVoyager  thanks！ Do you know how to  collect imu data and camera image synchronizely on android platform? I use api ”onPreviewFrame" to capture image, api "registerListener" and "onSensorChanged" to collect imu data. Does better way exist?

BITVoyager, 2018-01-08:
I think your way of logging images and sensor data is correct.

YZQ-Nine, 2018-02-28:
@wtmilk Hi,can you  share the android VINS project to me ?  I am trying to make it work on android as well. Thanks.
Here is my E-mail：17664080219@163.com.

QMonteCarlo, 2018-07-02:
Hi, @wtmilk ,you can try PilotGuru on Android phones with Android Version > 5.0. You can find the project both on github for source code or Google Play Store for apk. I also collect data on Xiaomi 6 with this app.
I'd like to find the reliability of the android phone sensors. Any feedback is welcome!