HKUST-Aerial-Robotics/VINS-Mono #145: Mobile Dataset

sinanariyurek, 2018-01-19:
I have read your papers E. Application II: Mobile Device section and saw that IMU data is in 100Hz and Image in 30Hz. From the Euroc and other datasets there were 10 imus inbetween, in iPhone you have 3-4 datas does not that cause any trouble. From your mobile code there is an interpolation of gyro to acc but no interpolation between frame and acc. Is it because they are synchronized? If you have stored your datas during iPhone test could you publish them? 
Thanks in advance

qintonguav, 2018-02-01:
We will release the bag collected by the mobile soon.

sinanariyurek, 2018-03-23:
Thanks it will be very helpful.

closed 2018-03-23 by sinanariyurek

shakith, 2018-07-04:
@sinanariyurek @qintonguav Was this iphone rosbag released? I am also interested to test VINS mono with mobile data and compare it with my mobile data. Thanks.

fangxu622, 2018-08-29:
hi~ my own devece captured the camera and imu data .how to make own rosbag data~have you some tourial?