HKUST-Aerial-Robotics/VINS-Mono #147: Equation 8 in ‘Technical Report'

liweizju, 2018-01-22:
Thanks for the great work and sharing the code!

I have a question with regard to the technical report attached with the project, that how equation (8) is derived? It says in the paper that "derive the continuous-time linearized dynamics of the error terms from eq4 and eq6", but what's the formulation of the error terms? 

Sorry if this is a stupid question, and I appreciate if anyone can give me hints!

tjuhuangshuai, 2018-01-23:
I think you can refer to the paper [ Quaternion kinematics for the error-state Kalman filter](http://www.iri.upc.edu/people/jsola/JoanSola/objectes/notes/kinematics.pdf)

liweizju, 2018-01-23:
@tjuhuangshuai Thanks for guidance~ Actually I have read this paper and I'm familiar with the derivatives of kinematics functions. I just...confused about what's the original formulation of the error term. Is it formed from equ(3) or ...else?

liweizju, 2018-01-23:
@tjuhuangshuai well... got it! Thanks!

closed 2018-01-23 by liweizju