HKUST-Aerial-Robotics/VINS-Mono #148: Loitorcam2pc

poudyalbot, 2018-01-23:
Dear sir,
Since you have mentioned you were able to run this using loitorcam2pc but there is no config file for the same sir is it possible to get good results with cheap sensors like loitorcam2pc for any tightly coupled visual inertial odometry.
Thank you

qintonguav, 2018-02-01:
This camera is out of stock, I can not get one. So we don't maintain this config.

You can check out to our previous version, which contains loitor config.
f5904b13c0ee649b6c1d4c7f3def8a228cb6dad7

poudyalbot, 2018-03-01:
When will your new paper for relocalization, map merging and global bundle
adjustment for visual inertial slam will be available for read.
Thank you

closed 2022-08-15 by poudyalbot