HKUST-Aerial-Robotics/VINS-Mono #149: Why there are two output?

zhaozhongch, 2018-01-24:
Hi thanks for your work!
I have a doubt that why we have two output?
One is called vins_"result_loop.csv" the other is called "result_no_loop.csv"
I just think for a SLAM system if we detect a loop then we have result_loop.csv if we didn't detect a loop then we have "result_no_loop.csv" I did't get it why you have two results together.
The loop here means something else??

Thanks for your answer!

Markbess, 2018-01-24:
I guess the system could choose to optimize the loop or not  when it detected a loop, .

qintonguav, 2018-02-01:
The one is visual-inertial odometry without loop closure. It's used for real-time control.
The other one is pose graph path under loop closure, pose graph optimization and drift correction. It's used for visualization.

zhaozhongch, 2018-02-01:
Thanks for your answer.
I got the visual-inertial odometry output
But the other on I am still confused. I mean the system doesn't tell me a loop is detected why it can have an output that has relationship loop closure? For example, like ORB-SLAM it detects a loop then it outputs the result after global bundle adjustment so the result should be like "result with loop closure".
But after I run VINS-Mono with Euroc dataset I didn't find any part it tell me "loop is detected" then is still has "result with loop closure"?

Thanks for your answer.

qintonguav, 2018-03-08:
If you see my video, you can find that I tell you the loop information by drawing loop links.

https://www.youtube.com/watch?v=eINyJHB34uU