HKUST-Aerial-Robotics/VINS-Mono #151: Nvidia jetson tx2 performance - delay in odometry response

mrsp, 2018-01-25:
Confirming it works on Jetson TX2 with:
- ubuntu 16.04
- Jetpack 3.1
- ros-kinetic
- Ceres 1.13
- Eigen **3.3.4** 

Hardware:
- mvBlueFOX-MLC200 camera (20fps) 
-xsens mti-300 imu (400Hz)

@20hz with loop closure, 50 features and rest parameters as euroc_config.yaml

Any tips on getting higher performance without compromising accuracy alot?
If i raise the camera fps e.g. to 30fps I notice a delay in the odometry response... 

Thanks in advance

qintonguav, 2018-02-01:
I recommend you run it with 10hz camera, 120 features.  A good balance between efficiency and accuracy.

mrsp, 2018-02-05:
Hello, I tried that and it seems to work more accurately (as expected).

Any tips on how to avoid filter's divergence when sudden accelerations happen (e.g. when hitting the ground)?

Thanks again

qintonguav, 2018-03-08:
set short camera exposure time.
set big IMU noise variance parameters.

mrsp, 2018-03-08:
I prefilterted the IMU measurements with a rolling median filter in order to detect sudden strikes as outliers and it worked.

By your experience could you tell me what exposure times (in us) to use in an outdoor/sunny environment?

Thanks

Kaapeine, 2022-03-17:
> I prefilterted the IMU measurements with a rolling median filter in order to detect sudden strikes as outliers and it worked.

Could you please provide some details of your implementation of this?