HKUST-Aerial-Robotics/VINS-Mono #154: Waiting for image and IMU data

Jdiesel666, 2018-01-31:
[ WARN] [1517358102.709370779]: have no prior about extrinsic param, calibrate extrinsic param
[ WARN] [1517358102.709473473]: waiting for image and imu...
[ WARN] [1517358102.731448542]: LOOP_CLOSURE true

My camera is publishing the data just fine when I echo it but the imu topic is not giving me anything.

Anyone has an idea?

indigomega, 2018-02-01:
Did you check your connections? Are you using USB or I2C? Which IMU are you using?

Jdiesel666, 2018-02-01:
@indigomega I'm using pixhawk with USB. I'm pretty sure my connection is solid.

indigomega, 2018-02-01:
Your question suggests to me that you can't get IMU data from pixhawk, so it's a problem with pixhawk, not with VINS. Am I correct? Try echoing other topics (try the barometer). If that doesn't work, check the BPS you set for your companion computer.

closed 2018-02-02 by Jdiesel666