HKUST-Aerial-Robotics/VINS-Mono #155: VINS on Robot

Cathy1987, 2018-02-08:
Hi, thanks for your great work!
Can VINS be used on robot, which moving at a constant speed usually. If IMU can't get the acceleration in this situation, what we can do?
Thank you for your time!

iGoSens, 2018-02-08:
With the same question, thank you very much.

qintonguav, 2018-03-08:
It's a general degraded case, I think monocular VINS cannot deal with it.

It's better  to solve it by fusing wheel odometry.

NikolausDemmel, 2018-04-16:
@jian-li: very helpful comment, thank you. Just minor typo I guess: `constant acceleration` --> `constant velocity`

jian-li, 2018-04-16:
Just is the constant acceleration, when moving in constant acceleration, you can not separate the linear acceleration from the acceleration bias. The constant velocity is one situation of the constant acceleration.

NikolausDemmel, 2018-04-16:
Ah ok, I wasn't really aware. Makes sense. Thanks for the clarification!

cathyyoung360, 2018-04-24:
Thanks a lot.

shengxingdong, 2018-06-27:
@jian-li   if intergrate stereo camera to initial feature points depth for optimization,  is it will helpful to reduce the problem(constant acceleration or static long time)?

jian-li, 2018-06-27:
For stereo camera, the scale is observable, so the problem does not exist.

shengxingdong, 2018-06-27:
@jian-li ，thank you.
so in the VINS code, if the depth of feature points are estimated by stereo camera instead of triangle by PNP and estimated by backend  optimization,  it is potential to solve the problem?

xiesc, 2018-08-22:
@jian-li 
I do not agree with " The constant velocity is one situation of the constant acceleration".
constant acceleration mean ak+1 = ak;  but if ak !=0, it will make vk changing.
so the bad situation should be constant velocity.

zhaoworking, 2020-12-20:
So, how to deal with this degenerated condition ？