HKUST-Aerial-Robotics/VINS-Mono #160: Question about Marginalization factor's residual

guofengw, 2018-02-28:
I found that in your MarginalizationFactor, the residual computing in the Evaluate function is like below:
Eigen::Map<Eigen::VectorXd>(residuals, n) = marginalization_info->linearized_residuals + marginalization_info->linearized_jacobians * dx;

But if i just set Eigen::Map<Eigen::VectorXd>(residuals, n) = dx, it is also OK!
Even set the residuals equal to 0 is also OK!

This confuse me a bit, so my question is that how is your residual figure out?  could you give some details about this part? Thank you!

qintonguav, 2018-03-08:
I guess you test it with EuRoC dataset.

In EuRoC, the measurements are of high quality. Even if you don't use marginalization factor, it can also achieve a good performance only in a local window without any history information.

If you use other general data, you may know the effect of marginalization factor.

stihuangyuan, 2018-04-18:
@qintonguav @guofengw  if existing parameter after marginalization global size not equal local size, when compute prior error, do we need to compute jacobian as linearized_jacobians * derivative local by global?