HKUST-Aerial-Robotics/VINS-Mono #162: catkin_make failed

phenix1983, 2018-03-02:
[  1%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/benchmark_publisher/benchmark_publisher
[ 21%] Built target camera_model
[ 23%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/camera_model/Calibration
CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o：在函数‘std::string readParam<std::string>(ros::NodeHandle&, std::string)’中：
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:21：对‘ros::NodeHandle::getParam(std::string const&, std::string&) const’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:27：对‘ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:23：对‘ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:23：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:27：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o：在函数‘void ros::Publisher::publish<nav_msgs::Odometry_<std::allocator<void> > >(nav_msgs::Odometry_<std::allocator<void> > const&) const’中：
/opt/ros/kinetic/include/ros/publisher.h:108：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/opt/ros/kinetic/include/ros/publisher.h:108：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/opt/ros/kinetic/include/ros/publisher.h:108：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o:/opt/ros/kinetic/include/ros/publisher.h:102: 跟着更多未定义的参考到 ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)
CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o：在函数‘main’中：
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:130：对‘ros::init(int&, char**, std::string const&, unsigned int)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:131：对‘ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:145：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:151：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:138：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o：在函数‘ros::SubscriptionCallbackHelperT<boost::shared_ptr<nav_msgs::Odometry_<std::allocator<void> > const> const&, void>::deserialize(ros::SubscriptionCallbackHelperDeserializeParams const&)’中：
/opt/ros/kinetic/include/ros/subscription_callback_helper.h:126：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
collect2: error: ld returned 1 exit status
VINS-Mono/benchmark_publisher/CMakeFiles/benchmark_publisher.dir/build.make:118: recipe for target '/home/yfz/catkin_ws/devel/lib/benchmark_publisher/benchmark_publisher' failed
make[2]: *** [/home/yfz/catkin_ws/devel/lib/benchmark_publisher/benchmark_publisher] Error 1
CMakeFiles/Makefile2:1830: recipe for target 'VINS-Mono/benchmark_publisher/CMakeFiles/benchmark_publisher.dir/all' failed
make[1]: *** [VINS-Mono/benchmark_publisher/CMakeFiles/benchmark_publisher.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[ 24%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/ar_demo/ar_demo_node
[ 26%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/vins_estimator/vins_estimator
CMakeFiles/Calibration.dir/src/intrinsic_calib.cc.o：在函数‘main’中：
intrinsic_calib.cc:(.text.startup+0xeb4)：对‘cv::imread(std::string const&, int)’未定义的引用
intrinsic_calib.cc:(.text.startup+0x1024)：对‘cv::imread(std::string const&, int)’未定义的引用
intrinsic_calib.cc:(.text.startup+0x1374)：对‘cv::_OutputArray::_OutputArray(cv::Mat&)’未定义的引用
intrinsic_calib.cc:(.text.startup+0x1392)：对‘cv::_InputArray::_InputArray(cv::Mat const&)’未定义的引用
intrinsic_calib.cc:(.text.startup+0x13b6)：对‘cv::imshow(std::string const&, cv::_InputArray const&)’未定义的引用
intrinsic_calib.cc:(.text.startup+0x15ad)：对‘cv::destroyWindow(std::string const&)’未定义的引用
intrinsic_calib.cc:(.text.startup+0x1832)：对‘cv::imread(std::string const&, int)’未定义的引用
intrinsic_calib.cc:(.text.startup+0x1a43)：对‘cv::putText(cv::Mat&, std::string const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool)’未定义的引用
intrinsic_calib.cc:(.text.startup+0x1a72)：对‘cv::_InputArray::_InputArray(cv::Mat const&)’未定义的引用
intrinsic_calib.cc:(.text.startup+0x1a96)：对‘cv::imshow(std::string const&, cv::_InputArray const&)’未定义的引用
CMakeFiles/Calibration.dir/src/calib/CameraCalibration.cc.o：在函数‘camodocal::CameraCalibration::drawResults(std::vector<cv::Mat, std::allocator<cv::Mat> >&) const’中：
CameraCalibration.cc:(.text+0x15fb)：对‘cv::circle(cv::Mat&, cv::Point_<int>, int, cv::Scalar_<double> const&, int, int, int)’未定义的引用
CameraCalibration.cc:(.text+0x1661)：对‘cv::circle(cv::Mat&, cv::Point_<int>, int, cv::Scalar_<double> const&, int, int, int)’未定义的引用
CameraCalibration.cc:(.text+0x1a45)：对‘cv::putText(cv::Mat&, std::string const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool)’未定义的引用
CameraCalibration.cc:(.text+0x1f0b)：对‘cv::_OutputArray::_OutputArray(cv::Mat&)’未定义的引用
CameraCalibration.cc:(.text+0x1f1b)：对‘cv::_InputArray::_InputArray(cv::Mat const&)’未定义的引用
CMakeFiles/Calibration.dir/src/calib/CameraCalibration.cc.o：在函数‘camodocal::CameraCalibration::optimize(boost::shared_ptr<camodocal::Camera>&, std::vector<cv::Mat, std::allocator<cv::Mat> >&, std::vector<cv::Mat, std::allocator<cv::Mat> >&) const’中：
CameraCalibration.cc:(.text+0x433b)：对‘cv::_OutputArray::_OutputArray(cv::Mat&)’未定义的引用
CameraCalibration.cc:(.text+0x434e)：对‘cv::_InputArray::_InputArray(cv::Mat const&)’未定义的引用
CMakeFiles/Calibration.dir/src/calib/CameraCalibration.cc.o：在函数‘void cv::cv2eigen<double, 3, 1, 0, 3, 1>(cv::Mat const&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&)’中：
CameraCalibration.cc:(.text._ZN2cv8cv2eigenIdLi3ELi1ELi0ELi3ELi1EEEvRKNS_3MatERN5Eigen6MatrixIT_XT0_EXT1_EXT2_EXT3_EXT4_EEE[_ZN2cv8cv2eigenIdLi3ELi1ELi0ELi3ELi1EEEvRKNS_3MatERN5Eigen6MatrixIT_XT0_EXT1_EXT2_EXT3_EXT4_EEE]+0xcd)：对‘cv::_OutputArray::_OutputArray(cv::Mat&)’未定义的引用
CameraCalibration.cc:(.text._ZN2cv8cv2eigenIdLi3ELi1ELi0ELi3ELi1EEEvRKNS_3MatERN5Eigen6MatrixIT_XT0_EXT1_EXT2_EXT3_EXT4_EEE[_ZN2cv8cv2eigenIdLi3ELi1ELi0ELi3ELi1EEEvRKNS_3MatERN5Eigen6MatrixIT_XT0_EXT1_EXT2_EXT3_EXT4_EEE]+0x4df)：对‘cv::_OutputArray::_OutputArray(cv::Mat&)’未定义的引用
CameraCalibration.cc:(.text._ZN2cv8cv2eigenIdLi3ELi1ELi0ELi3ELi1EEEvRKNS_3MatERN5Eigen6MatrixIT_XT0_EXT1_EXT2_EXT3_EXT4_EEE[_ZN2cv8cv2eigenIdLi3ELi1ELi0ELi3ELi1EEEvRKNS_3MatERN5Eigen6MatrixIT_XT0_EXT1_EXT2_EXT3_EXT4_EEE]+0x506)：对‘cv::_OutputArray::_OutputArray(cv::Mat&)’未定义的引用
CameraCalibration.cc:(.text._ZN2cv8cv2eigenIdLi3ELi1ELi0ELi3ELi1EEEvRKNS_3MatERN5Eigen6MatrixIT_XT0_EXT1_EXT2_EXT3_EXT4_EEE[_ZN2cv8cv2eigenIdLi3ELi1ELi0ELi3ELi1EEEvRKNS_3MatERN5Eigen6MatrixIT_XT0_EXT1_EXT2_EXT3_EXT4_EEE]+0x519)：对‘cv::_InputArray::_InputArray(cv::Mat const&)’未定义的引用
[... cut ...]
make[1]: *** [VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Base path: /home/yfz/catkin_ws
Source space: /home/yfz/catkin_ws/src
Build space: /home/yfz/catkin_ws/build
Devel space: /home/yfz/catkin_ws/devel
Install space: /home/yfz/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/yfz/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/yfz/catkin_ws/build"
####
Invoking "make -j4 -l4" failed

phenix1983, 2018-03-02:
I just run "catkin_make" at ~/catkin_ws ,then encounter the problem. Thanks in advance.:)

phenix1983, 2018-03-02:
after refreshed openCV3.3.1, 
 the error is like this:

[  1%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/benchmark_publisher/benchmark_publisher
[ 21%] Built target camera_model
CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o：在函数‘std::string readParam<std::string>(ros::NodeHandle&, std::string)’中：
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:21：对‘ros::NodeHandle::getParam(std::string const&, std::string&) const’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:27：对‘ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:23：对‘ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:23：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:27：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o：在函数‘void ros::Publisher::publish<nav_msgs::Odometry_<std::allocator<void> > >(nav_msgs::Odometry_<std::allocator<void> > const&) const’中：
/opt/ros/kinetic/include/ros/publisher.h:108：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/opt/ros/kinetic/include/ros/publisher.h:108：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/opt/ros/kinetic/include/ros/publisher.h:108：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o:/opt/ros/kinetic/include/ros/publisher.h:102: 跟着更多未定义的参考到 ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)
CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o：在函数‘main’中：
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:130：对‘ros::init(int&, char**, std::string const&, unsigned int)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:131：对‘ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:145：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:151：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/benchmark_publisher/src/benchmark_publisher_node.cpp:138：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o：在函数‘ros::SubscriptionCallbackHelperT<boost::shared_ptr<nav_msgs::Odometry_<std::allocator<void> > const> const&, void>::deserialize(ros::SubscriptionCallbackHelperDeserializeParams const&)’中：
/opt/ros/kinetic/include/ros/subscription_callback_helper.h:126：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
collect2: error: ld returned 1 exit status
VINS-Mono/benchmark_publisher/CMakeFiles/benchmark_publisher.dir/build.make:118: recipe for target '/home/yfz/catkin_ws/devel/lib/benchmark_publisher/benchmark_publisher' failed
make[2]: *** [/home/yfz/catkin_ws/devel/lib/benchmark_publisher/benchmark_publisher] Error 1
CMakeFiles/Makefile2:1830: recipe for target 'VINS-Mono/benchmark_publisher/CMakeFiles/benchmark_publisher.dir/all' failed
make[1]: *** [VINS-Mono/benchmark_publisher/CMakeFiles/benchmark_publisher.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[ 23%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/camera_model/Calibration
[ 24%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/ar_demo/ar_demo_node
[ 26%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/vins_estimator/vins_estimator
CMakeFiles/Calibration.dir/src/camera_models/CameraFactory.cc.o：在函数‘camodocal::CameraFactory::generateCameraFromYamlFile(std::string const&)’中：
CameraFactory.cc:(.text+0xe50)：对‘cv::read(cv::FileNode const&, std::string&, std::string const&)’未定义的引用
CMakeFiles/Calibration.dir/src/camera_models/PinholeCamera.cc.o：在函数‘camodocal::PinholeCamera::Parameters::readFromYamlFile(std::string const&)’中：
PinholeCamera.cc:(.text+0x56ce)：对‘cv::read(cv::FileNode const&, std::string&, std::string const&)’未定义的引用
PinholeCamera.cc:(.text+0x5970)：对‘cv::read(cv::FileNode const&, std::string&, std::string const&)’未定义的引用
[... cut ...]
make[1]: *** [VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Base path: /home/yfz/catkin_ws
Source space: /home/yfz/catkin_ws/src
Build space: /home/yfz/catkin_ws/build
Devel space: /home/yfz/catkin_ws/devel
Install space: /home/yfz/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/yfz/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/yfz/catkin_ws/build"
####
Invoking "make -j4 -l4" failed

zhaoworking, 2021-03-05:
so,have you solved the problem, cause I also encountered this trouble.

closed 2024-02-25 by qintonguav