HKUST-Aerial-Robotics/VINS-Mono #163: Failed to catkin_make.

phenix1983, 2018-03-03:
Hi, 
     I think iI have installed all the dependencies,  when run catkin_make, errors are as follows:
yfz@yfz-ZHAOYANG-E42-80:~/catkin_ws$ catkin_make
Base path: /home/yfz/catkin_ws
Source space: /home/yfz/catkin_ws/src
Build space: /home/yfz/catkin_ws/build
Devel space: /home/yfz/catkin_ws/devel
Install space: /home/yfz/catkin_ws/install
####
#### Running command: "cmake /home/yfz/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/yfz/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/yfz/catkin_ws/install -G Unix Makefiles" in "/home/yfz/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/yfz/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/yfz/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/yfz/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/yfz/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.8
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - beginner_tutorials
-- ~~  - benchmark_publisher
-- ~~  - camera_model
-- ~~  - ar_demo
-- ~~  - feature_tracker
-- ~~  - pose_graph
-- ~~  - vins_estimator
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'beginner_tutorials'
-- ==> add_subdirectory(beginner_tutorials)
-- +++ processing catkin package: 'benchmark_publisher'
-- ==> add_subdirectory(VINS-Mono/benchmark_publisher)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'camera_model'
-- ==> add_subdirectory(VINS-Mono/camera_model)
CMake Warning at /usr/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106600
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:763 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:1332 (_Boost_MISSING_DEPENDENCIES)
  VINS-Mono/camera_model/CMakeLists.txt:13 (find_package)

CMake Warning at /usr/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106600
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:763 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:1332 (_Boost_MISSING_DEPENDENCIES)
  VINS-Mono/camera_model/CMakeLists.txt:13 (find_package)

CMake Warning at /usr/share/cmake-3.5/Modules/FindBoost.cmake:725 (message):
  Imported targets not available for Boost version 106600
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:763 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.5/Modules/FindBoost.cmake:1332 (_Boost_MISSING_DEPENDENCIES)
  VINS-Mono/camera_model/CMakeLists.txt:13 (find_package)

-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   filesystem
--   program_options
--   system
-- Found OpenCV: /opt/ros/kinetic (found version "3.3.1") 
-- Found installed version of Eigen: /usr/local/share/eigen3/cmake
-- Found required Ceres dependency: Eigen version 3.3.3 in /usr/local/include/eigen3
[... 210 lines cut ...]
CMakeFiles/vins_estimator.dir/src/utility/visualization.cpp.o：在函数‘void ros::Publisher::publish<nav_msgs::Odometry_<std::allocator<void> > >(nav_msgs::Odometry_<std::allocator<void> > const&) const’中：
/opt/ros/kinetic/include/ros/publisher.h:108：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/opt/ros/kinetic/include/ros/publisher.h:108：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
CMakeFiles/vins_estimator.dir/src/utility/visualization.cpp.o:/opt/ros/kinetic/include/ros/publisher.h:108: 跟着更多未定义的参考到 ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)
CMakeFiles/vins_estimator.dir/src/initial/initial_aligment.cpp.o：在函数‘LinearAlignment(std::map<double, ImageFrame, std::less<double>, std::allocator<std::pair<double const, ImageFrame> > >&, Eigen::Matrix<double, 3, 1, 0, 3, 1>&, Eigen::Matrix<double, -1, 1, 0, -1, 1>&)’中：
/home/yfz/catkin_ws/src/VINS-Mono/vins_estimator/src/initial/initial_aligment.cpp:192：对‘ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/vins_estimator/src/initial/initial_aligment.cpp:192：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/vins_estimator/src/initial/initial_aligment.cpp:183：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/vins_estimator/src/initial/initial_aligment.cpp:181：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/vins_estimator/src/initial/initial_aligment.cpp:183：对‘ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)’未定义的引用
CMakeFiles/vins_estimator.dir/src/initial/initial_aligment.cpp.o：在函数‘solveGyroscopeBias(std::map<double, ImageFrame, std::less<double>, std::allocator<std::pair<double const, ImageFrame> > >&, Eigen::Matrix<double, 3, 1, 0, 3, 1>*)’中：
/home/yfz/catkin_ws/src/VINS-Mono/vins_estimator/src/initial/initial_aligment.cpp:27：对‘ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)’未定义的引用
/home/yfz/catkin_ws/src/VINS-Mono/vins_estimator/src/initial/initial_aligment.cpp:27：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
CMakeFiles/vins_estimator.dir/src/initial/initial_ex_rotation.cpp.o：在函数‘InitialEXRotation::testTriangulation(std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&, std::vector<cv::Point_<float>, std::allocator<cv::Point_<float> > > const&, cv::Mat_<double>, cv::Mat_<double>)’中：
/home/yfz/catkin_ws/src/VINS-Mono/vins_estimator/src/initial/initial_ex_rotation.cpp:123：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
CMakeFiles/vins_estimator.dir/src/initial/initial_ex_rotation.cpp.o：在函数‘InitialEXRotation::CalibrationExRotation(std::vector<std::pair<Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1> >, std::allocator<std::pair<Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1> > > >, Eigen::Quaternion<double, 0>, Eigen::Matrix<double, 3, 3, 0, 3, 3>&)’中：
/home/yfz/catkin_ws/src/VINS-Mono/vins_estimator/src/initial/initial_ex_rotation.cpp:27：对‘ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)’未定义的引用
collect2: error: ld returned 1 exit status
VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/build.make:554: recipe for target '/home/yfz/catkin_ws/devel/lib/vins_estimator/vins_estimator' failed
make[2]: *** [/home/yfz/catkin_ws/devel/lib/vins_estimator/vins_estimator] Error 1
CMakeFiles/Makefile2:2309: recipe for target 'VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/all' failed
make[1]: *** [VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

safijari, 2019-07-03:
I had this issue when I was using catkin tools and had the workspace set up to make an install space. I had to make a new devel space workspace to build VINS

Mrxiang-Plus, 2019-07-07:
i have meet this program too ,i don't know how to deal with it

FrancescoFallica, 2020-04-30:
Try to remove build, devel, and logs folder and compile with `catkin build`