HKUST-Aerial-Robotics/VINS-Mono #164: Catkin_make failed

phenix1983, 2018-03-06:
[  1%] Building CXX object VINS-Mono/benchmark_publisher/CMakeFiles/benchmark_publisher.dir/src/benchmark_publisher_node.cpp.o
[  3%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/chessboard/Chessboard.cc.o
[  4%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/intrinsic_calib.cc.o
[  6%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator_node.cpp.o
[  7%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/calib/CameraCalibration.cc.o
[  9%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/chessboard/Chessboard.cc.o
[ 10%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/benchmark_publisher/benchmark_publisher
[ 10%] Built target benchmark_publisher
[ 12%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/calib/CameraCalibration.cc.o
[ 13%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/Camera.cc.o
[ 15%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/camera_models/Camera.cc.o
[ 16%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/CameraFactory.cc.o
[ 18%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/camera_models/CameraFactory.cc.o
[ 20%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/camera_models/CostFunctionFactory.cc.o
[ 21%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/camera_models/PinholeCamera.cc.o
[ 23%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/CostFunctionFactory.cc.o
[ 24%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/camera_models/CataCamera.cc.o
[ 26%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/camera_models/EquidistantCamera.cc.o
[ 27%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/parameters.cpp.o
[ 29%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/camera_models/ScaramuzzaCamera.cc.o
[ 30%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/estimator.cpp.o
[ 32%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/sparse_graph/Transform.cc.o
[ 33%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/gpl/gpl.cc.o
[ 35%] Building CXX object VINS-Mono/camera_model/CMakeFiles/Calibration.dir/src/gpl/EigenQuaternionParameterization.cc.o
[ 36%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/PinholeCamera.cc.o
[ 38%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/CataCamera.cc.o
[ 40%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/EquidistantCamera.cc.o
[ 44%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/camera_model/Calibration
[ 44%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/feature_manager.cpp.o
[ 44%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/camera_models/ScaramuzzaCamera.cc.o
[ 46%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/sparse_graph/Transform.cc.o
[ 47%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/factor/pose_local_parameterization.cpp.o
[ 49%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/factor/projection_factor.cpp.o
[ 50%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/gpl/gpl.cc.o
[ 52%] Building CXX object VINS-Mono/camera_model/CMakeFiles/camera_model.dir/src/gpl/EigenQuaternionParameterization.cc.o
[ 53%] Linking CXX shared library /home/yfz/catkin_ws/devel/lib/libcamera_model.so
CMakeFiles/Calibration.dir/src/intrinsic_calib.cc.o：在函数‘boost::program_options::typed_value<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, char>::xparse(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) const’中：
intrinsic_calib.cc:(.text._ZNK5boost15program_options11typed_valueINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEcE6xparseERNS_3anyERKSt6vectorIS7_SaIS7_EE[_ZNK5boost15program_options11typed_valueINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEcE6xparseERNS_3anyERKSt6vectorIS7_SaIS7_EE]+0x17)：对‘boost::program_options::validate(boost::any&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, int)’未定义的引用
CMakeFiles/Calibration.dir/src/intrinsic_calib.cc.o：在函数‘boost::program_options::validation_error::validation_error(boost::program_options::validation_error::kind_t, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)’中：
intrinsic_calib.cc:(.text._ZN5boost15program_options16validation_errorC2ENS1_6kind_tERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESA_i[_ZN5boost15program_options16validation_errorC5ENS1_6kind_tERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESA_i]+0x36)：对‘boost::program_options::validation_error::get_template[abi:cxx11](boost::program_options::validation_error::kind_t)’未定义的引用
intrinsic_calib.cc:(.text._ZN5boost15program_options16validation_errorC2ENS1_6kind_tERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESA_i[_ZN5boost15program_options16validation_errorC5ENS1_6kind_tERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESA_i]+0x4c)：对‘boost::program_options::error_with_option_name::error_with_option_name(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)’未定义的引用
CMakeFiles/Calibration.dir/src/intrinsic_calib.cc.o：在函数‘boost::program_options::basic_command_line_parser<char>::basic_command_line_parser(int, char const* const*)’中：
intrinsic_calib.cc:(.text._ZN5boost15program_options25basic_command_line_parserIcEC2EiPKPKc[_ZN5boost15program_options25basic_command_line_parserIcEC5EiPKPKc]+0x2cc)：对‘boost::program_options::to_internal(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)’未定义的引用
intrinsic_calib.cc:(.text._ZN5boost15program_options25basic_command_line_parserIcEC2EiPKPKc[_ZN5boost15program_options25basic_command_line_parserIcEC5EiPKPKc]+0x340)：对‘boost::program_options::detail::cmdline::cmdline(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)’未定义的引用
CMakeFiles/Calibration.dir/src/intrinsic_calib.cc.o：在函数‘boost::program_options::typed_value<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, char>::name() const’中：
intrinsic_calib.cc:(.text._ZNK5boost15program_options11typed_valueINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEcE4nameEv[_ZNK5boost15program_options11typed_valueINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEcE4nameEv]+0x38)：对‘boost::program_options::arg[abi:cxx11]’未定义的引用
CMakeFiles/Calibration.dir/src/intrinsic_calib.cc.o：在函数‘boost::program_options::typed_value<float, char>::name() const’中：
intrinsic_calib.cc:(.text._ZNK5boost15program_options11typed_valueIfcE4nameEv[_ZNK5boost15program_options11typed_valueIfcE4nameEv]+0x38)：对‘boost::program_options::arg[abi:cxx11]’未定义的引用
CMakeFiles/Calibration.dir/src/intrinsic_calib.cc.o：在函数‘boost::program_options::typed_value<int, char>::name() const’中：
intrinsic_calib.cc:(.text._ZNK5boost15program_options11typed_valueIicE4nameEv[_ZNK5boost15program_options11typed_valueIicE4nameEv]+0x38)：对‘boost::program_options::arg[abi:cxx11]’未定义的引用
CMakeFiles/Calibration.dir/src/intrinsic_calib.cc.o：在函数‘main’中：
[... cut ...]
[ 64%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/initial/initial_sfm.cpp.o
[ 66%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/initial/initial_aligment.cpp.o
[ 67%] Building CXX object VINS-Mono/vins_estimator/CMakeFiles/vins_estimator.dir/src/initial/initial_ex_rotation.cpp.o
[ 69%] Linking CXX executable /home/yfz/catkin_ws/devel/lib/vins_estimator/vins_estimator
[ 69%] Built target vins_estimator
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

MerlotQ, 2018-03-08:
you may have to change the version of gcc like using gcc 5.4 etc

phenix1983, 2018-03-09:
Thank you for your reply. I tried to resolve the problem be reinstall gcc5.4.1, installation is successful, But the problem still exist. So, I use a new ubuntu16.04 computer, The error information dispeared.  The issue can be closed. The same as ISSUE99. Thank you. :)

cmshinde, 2019-10-11:
Facing the same error. I tried to do catkin_make using gcc5.4.1 also but no success