HKUST-Aerial-Robotics/VINS-Mono #165: Initialization is highly random

hngenc, 2018-03-07:
I created my own dataset using the [AirSim](https://github.com/Microsoft/AirSim) drone simulator. You can find the rosbag file [here](https://drive.google.com/file/d/186yTpdTW4ovwWOlrwAHfwT9uXdZAd9lx/view?usp=sharing), and the yaml file [here](https://drive.google.com/file/d/1cpO2XQSek-MVM7xSbQioHBYg0soKPlNi/view?usp=sharing).

VINS-Mono performance varies _drastically_ depending on where in the rosbag file we start playing from:
- If we start the rosbag file from the beginning, VINS-Mono fails to initialize the drone's position, and it never fully corrects itself.
- If we start one second into the rosbag file, VINS-Mono initializes well and does well over the entire run.
- If start 6 seconds into the rosbag file, VINS-Mono again fails to initialize its position correctly, resulting in a very inaccurate trajectory.
- But if we start 5 seconds into the rosbag file, VINS-Mono does great!

I'm really struggling to figure out why it is that VINS-Mono's performance seems to depend so heavily on the exact second that it starts running. I suspect that AirSim's simulated sensors may be a little too inaccurate, but that doesn't explain why VINS-Mono has no problem if it starts 1 second into my rosbag, but does so much worse if it starts 0 seconds into my rosbag file.

To help visualize, this is how it looks when VINS-Mono is initialized poorly (the drone was standing still while VINS-Mono's estimated trajectory shot off into the distance):

qintonguav, 2018-03-08:
It's mainly because of your movement pattern. It's a typically degraded case for the monocular camera.
I see your simulated video, the aircraft do the exactly rotation-only movement without any translation in some certain period.
As for a monocular camera, it's well known that you can't triangulate depth without translation. 
So that's why the initialization works badly in the rotation-only movement. 
It's a common issue in the most monocular algorithm.
I suggest you add some translation motion in the beginning.

Beniko95J, 2020-03-04:
@hngenc Hi, I am also trying to run VINS-Mono on airsim, and may I ask how did you get the extrinsic parameters between camera and IMU, and did you just use the timestamp from airsim directly other than modifying them? ([Here](https://github.com/microsoft/AirSim/issues/2369) is a discussion about the **soft time stamp** and **simulation time stamp** in airsim)

Thanks!

hngenc, 2020-12-13:
@Beniko95J This was quite a while ago, and my memory is pretty hazy now. IIRC, you can open up the model directly in Unreal Engine in order to see how far apart the camera are placed from each other, and what their rotation is. We just used the AirSim time stamps; I don't remember a distinction between two different time stamps when we were working on this, but maybe I just forgot.

And I see that I left this issue open for far too long, so a very late thank you for your answer, @qintonguav :)

closed 2020-12-13 by hngenc

ssaigarimella, 2025-07-04:
@hngenc Thanks for posting this question, was wondering the same. Would you be able to share the yaml config please? The link in the original post seems to have expired.