HKUST-Aerial-Robotics/VINS-Mono #166: Realsense Support

shakith, 2018-03-08:
Hi,

I see that real sense is supported by VINS-mono. Which model is that?

Where does the performance of the real sense fit in the following scale?

(global shutter camera + synchronized high-end IMU, e.g. VI-Sensor) > (global shutter camera + synchronized low-end IMU) > (global camera + unsync high frequency IMU) > (global camera + unsync low frequency IMU) > (rolling camera + unsync low frequency IMU).

Thanks,

ealdaz, 2018-04-16:
jian-li could you resend your link? It seems broken and I'm very interested in your results, thanks

jian-li, 2018-04-16:
If you copy the link on the browser, it should be ok. If it is still failed, maybe you should use a proxy.

ealdaz, 2018-04-19:
Copying the link worked, clicking on it doesn't, thanks !