HKUST-Aerial-Robotics/VINS-Mono #167: How cam i get the IMU noise parameter

derekhe89, 2018-03-10:
Hello,  i got the IMU noise parameter use allan deviation, but it didn't work well, can you tell me what's your method to get the IMU noise parameter? Thank you !

pxlong, 2018-03-10:
I have the same issue. @qintonguav 
And I found that the default IMU noise parameters worked for most cases.

PeiliangLi, 2018-03-10:
Try this repo: https://github.com/gaowenliang/imu_utils

xiaofeifei1234, 2018-03-15:
Hello ,i have suffered the same problem about  imu parameters, have you got the right parameters with  https://github.com/gaowenliang/imu_utils,thankyou @derekhe89

pxlong, 2018-03-15:
I have tried imu_utils.
I think you may get the right parameters with that repo. But it does NOT mean that the parameters are suitable for VINS-Mono.

xiaofeifei1234, 2018-03-15:
ok ,got it thank you for your reply,i have done the same work with this repo https://github.com/aewallin/allantools,but the parameters  didn't work  well, do you have any good idea to get suitable parameters for VINS-Mono?

pxlong, 2018-03-15:
check the IMU parameters in the config files, you can find the typical value of these parameters.
But It will be better if we can get the "correct" answer from authors @PeiliangLi  @qintonguav 
Thanks.

xiaofeifei1234, 2018-03-18:
yeah ,thank you i always use  the default parameters, hope we could get the better answer from the authors@ @PeiliangLi @dvorak0

csroccv, 2018-09-12:
in the file euroc_config.yaml,note like:
acc_n: 0.2 # accelerometer measurement noise standard deviation. #0.2
gyr_n: 0.02 # gyroscope measurement noise standard deviation. #0.05
acc_w: 0.0002 # accelerometer bias random work noise standard deviation. #0.02
gyr_w: 2.0e-5 # gyroscope bias random work noise standard deviation. #4.0e-5

but I feel they can't calculate by allen variance,by imu_utils (the tool calculate allen variance).What exactly does they mean?How to get them?How to calculate them?