HKUST-Aerial-Robotics/VINS-Mono #172: VINS使用Tara相机采集数据位姿漂移

Cathy1987, 2018-03-14:
您好！我使用Tara-stereo相机(只使用左目和IMU)采集数据，IMU 型号为LSM6DS0，输出频率为140Hz，camera是全局快门，网盘中数据集输出频率为10FPS，相机在小范围内运动，但运行VINS时会产生很大的漂移，如下图所示：
网盘中附有我们采集的数据包和相机配置文件，可以直接跑VINS，能否帮忙分析一下这可能是什么原因导致的呢，非常感谢！
 
链接: https://pan.baidu.com/s/15ZIaZ4EqHVuGFctOelrHMA 密码: aam7
其中 imu topic:  /imu0  camera topic: /cam0/image_raw

[image]

cathyyoung360, 2018-03-15:
We doubt that the IMU parameters may have some effects, so we calibrate it using the following utilities:
https://github.com/gaowenliang/imu_utils
Then we substitute the IMU parameters in tara_euroc_config.yaml for the calibrated result. However, the position still drifts. 

%YAML:1.0
---
type: IMU
name: tara
Gyr:
   unit: " rad/s"
   avg-axis:
      gyr_n: 1.7577553183142076e-04
      gyr_w: 1.2610317851150028e-04
   x-axis:
      gyr_n: 2.2804423561804073e-04
      gyr_w: 1.8398466458921368e-04
   y-axis:
      gyr_n: 1.7421737699998231e-04
      gyr_w: 1.1144172953448703e-04
   z-axis:
      gyr_n: 1.2506498287623925e-04
      gyr_w: 8.2883141410800115e-05
Acc:
   unit: " m/s^2"
   avg-axis:
      acc_n: 6.4956434654556193e-04
      acc_w: 5.6763275224198148e-04
   x-axis:
      acc_n: 6.2226036199732396e-04
      acc_w: 5.0519253057808224e-04
   y-axis:
      acc_n: 5.2240487931964528e-04
      acc_w: 4.6224038955112990e-04
   z-axis:
      acc_n: 8.0402779831971677e-04
      acc_w: 7.3546533659673218e-04

cathyyoung360, 2018-03-19:
Oh, it fixed. But I still wonder how can I calibrate the IMU noise and random walk effectively.Thank you!

Garfield-kh, 2018-04-23:
@cathyyoung360 I have a similar drift problem using another camera. May I know how do you fix the drift problem? Thank you very much.

cathyyoung360, 2018-04-24:
You can check out if your camera is well calibrated and whether the camera and IMU is synchronized.

wangbisheng, 2018-06-07:
My camera is 25Hz , and IMU is about 98Hz, so it is hard to keep them synchronizing always.  my camera is well calibrated ,  I have a similar drift problem. What 's main factor for this problem? What can i do for this problem? thank you!

cathyyang666, 2018-06-10:
You can set the parameter "estimate_td" to 1. It will optimize the time offset by itself. But the drift may caused by a lot of reasons, such as there are little features, moving too fast , or the IMU is poor quality.

LiuSQ123, 2018-08-31:
@wangbisheng ,Hello,do you fix this problem??How to synchronize the IMU and Camera data??The system on my own device work bad.............