HKUST-Aerial-Robotics/VINS-Mono #173: triangulate points come out negative depth value

Maczzz, 2018-03-14:
Hi,
When triangulate points, sometimes a few of points come out negative depth value.
i see from debug data, the feature tracked point  is good.
so, I want to know the reason for this problem. [pure rotate? image rectification not good? or others?]

` negative depth data`
 `mocular size=4 origin x/z=0.231535 y/z=-0.347558 z/w=-2.774240 w=-0.315618`
`[frame i]  translation norm=0.000000 x=0.230027 y=-0.342712 z=1.000000 u=520.000000 v=74.000000`
`[frame i+1]  translation norm=0.013260 x=0.171813 y=-0.341118 z=1.000000 u=491.273346 v=73.109055`
`[frame i+2]  translation norm=0.021704 x=0.136717 y=-0.340838 z=1.000000 u=473.576965 v=72.470230`
`[frame i+3]  translation norm=0.027971 x=0.126995 y=-0.341761 z=1.000000 u=468.617523 v=71.855774`

`positive depth data`
`mocular size=4 origin x/z=0.112430 y/z=-0.345289 z/w=5.176776 w=0.178649`
`[frame i]  translation norm=0.000000 x=0.113909 y=-0.345341 z=1.000000 u=461.883301 v=69.957207`
`[frame i+1]   translation norm=0.029094 x=0.097417 y=-0.348330 z=1.000000 u=453.402252 v=68.310539`
`[frame i+2]  translation norm=0.044186 x=0.089816 y=-0.349068 z=1.000000 u=449.495331 v=67.862045`
`[frame i+3]  translation norm=0.060465 x=0.083699 y=-0.351784 z=1.000000 u=446.321014 v=66.520966`

[x,y,z] is camera unit plane coordinate , [u,v] is pixel coordinate, [x/z, y/z] is 3D world point Projection  camera unit plane coordinate. z/w is 3d point depth value.

If someone could help, i will be deeply thankful.
with the best regards

stavBodik, 2019-06-19:
@Maczzz  same here ):

Sarra776, 2022-06-28:
@Maczzz @stavBodik, have you already fixed the problem?
I also have negative outputs...