HKUST-Aerial-Robotics/VINS-Mono #184: Issues with Loitor camera

indigomega, 2018-04-06:
Hey, I'm having significant issues running VINS with my loitor camera. I keep getting drifting odometry and a misalign in visual structure with IMU warning. I've calibrated the camera multiple times as pinhole and MEI (with a 0.2 pixel reprojection error) and I've tried revising the driver to see if there are any synchronization issues. I've also manually installed Eigen 3.3.3 on my system (16.04). I've even calculated my IMU's parameters using the imu_utils package. I've tried playing around with camera settings but to no avail. Any ideas? I'm clueless at this point.

indigomega, 2018-04-10:
Well I've given up. To anyone who stumbles upon this make note: ensure that your hardware is well synchronized and works properly. VINS has strict requirements.

closed 2018-04-10 by indigomega

YiChenCityU, 2018-04-17:
try use Boteye or ZR300， loitor is unstable.