HKUST-Aerial-Robotics/VINS-Mono #187: IMU parameters

duyilong, 2018-04-11:
with self-made VI equipment(like camera + mpu6050module)，IMU paremeters are not easy to get.
Is that important?

IbaDup, 2018-04-11:
If your IMU & Camera is synchronized, no they are not. Assuming that you are talking about the noise and random walk of bias. These parameters are used to find a better middle ground between your IMU and camera. If your IMU has large noise the system depends more on the camera. You can start with high values. Also in VINS the noise is assumed higher than given in the dataset.

duyilong, 2018-04-11:
@IbaDup Thanks for replying. As usual, how higher the value could be? like 10 times of the default vlues in project config flie?(acc_n: 0.2  gyr_n: 0.02 acc_w: 0.0002 gyr_w: 2.0e-5 ),is that feasible, I hope.
Anyway, your explanation is very useful to me, thanks so much.

csroccv, 2018-09-12:
What exactly does acc_n、 gyr_n、 acc_w、 gyr_w mean? I feel they  can't calculate by allen variance.How to get them?How to calculate them?@duyilong@lbaDup

duyilong, 2018-09-12:
Well, I didn't get those parameters by allen variance either. @csroccv