HKUST-Aerial-Robotics/VINS-Mono #192: Create mask for fisheye camera

tuandle, 2018-04-27:
Hi, 
I would like to ask how to draw a mask for fisheye camera. You briefly mentioned it in #26 but not very detail. 
What do you mean by "draw a circle mask in the same size" ?
Let's take your fisheye_mask for example. Do you mean the white circle needs to be the same as image size?

tuandle, 2018-05-08:
@qintonguav Could you help me with this?

qintonguav, 2018-05-08:
what's resolution of your camera?

tuandle, 2018-05-08:
I use [ironsides](https://www.perceptin.io/ironsides) camera. Its resolution is 640x480.

qintonguav, 2018-05-08:
[image]

tuandle, 2018-05-08:
@qintonguav thank you so much for your help. May I ask how can you draw this mask? Is there any source that you can point me to? Thanks a gain!

qintonguav, 2018-05-08:
1 #include <opencv2/opencv.hpp>
  2 using namespace std;
  3 using namespace cv;
  4 
  5 int main()
  6 {
  7     Mat image = Mat::zeros(480, 640, CV_8UC1 );
  8     circle(image, cv::Point2f(320, 240), 320, Scalar(255), -1, 8, 0);
  9     imwrite("mask.jpg", image);
 10     imshow("circle", image);
 11     waitKey(0);
 12     return 0;
 13 }

complile and run.

tuandle, 2018-05-08:
Great! Thanks!

closed 2018-05-08 by tuandle