HKUST-Aerial-Robotics/VINS-Mono #195: Using loitor with VINS-Mono

shakith, 2018-05-07:
Hi @qintonguav,

I am trying to use the loitor camera+IMU with VINS-Mono. I have also tried VINS mobile on the iphone. Compared to VINS mobile, VINS-Mono is unstable with drift and tracking is not robust. I see sometimes td of 3-6ms even though there is unstable drift.

Is this expected with the two different hardware in the iphone and the loitor? Loitor would have higher performance in the following order.

(global shutter camera + synchronized high-end IMU, e.g. VI-Sensor) > (global shutter camera + synchronized low-end IMU) **LOITOR**  > (global camera + unsync high frequency IMU) > (global camera + unsync low frequency IMU) > (rolling camera + unsync low frequency IMU) **IPHONE**.

What is the SDK code for the loitor you have tested with  VINS-Mono? My loitor/optor sdk is https://github.com/optor-vis/vi-stereo-v1
What is the setup file for loitor settings used with VINS-Mono ? e.g., https://github.com/optor-vis/vi-stereo-v1/blob/master/optor_VI_Sensor_SDK_V1.0/ROS/optor_stereo_visensor/optor_VISensor_Setups.txt
There are frequent frame drops from loitor. Did you customize the loitor for VINS?

Have other people used loitor successfully and robustly?
Alternatively, what are the supported camera + IMU hardware for VINS?

maciejmatuszak, 2018-05-09:
Yes, the Loitor has plenty of problems:
- frame drops: (white frames)
- duplicate frames with 1us difference in timestamp
- lack of technical support
It was long time since I play with VINS-Mono, from memory as long as there was excitation in the IMU the position was tracked ok. When I would put the UAV down it would evently drift. 
I modified Loitor SDK you can check the miner-uav branch: https://github.com/maciejmatuszak/vi_sensor_sdk/tree/miner-uav 
I did not addressed the frame drops or duplicates yet as those are not critical in what I am using for now. 
I will have to address the duplicates as the ROVIOLI framework that I am using is complaining that there is no IMU data between frames and crushes. I think the duplicate is artefact of the synchronisation method they use but it need to be investigated further.
In the branch mentioned above you will find:
- support for camera_info messages and service (you can use the ros stereo camera calibration easly)
- image compression (image_transport) support
- automatic IMU acceleration scaling when IMU is not moving. (It will scale it to G value computed from latitude and asl height)
- IMU data rotation

Thinking back about VINS-Mono it is optimisation based SLAM and it need enough CPU time per frame to get good solution. If other tasks are running those can still the CPU time and cause optimisation to timeout and the tracking to diverge quickly. I think setting taskset command in roslaunch (https://answers.ros.org/question/202712/how-to-bind-a-node-to-a-specific-cpu-core/ ) scripts for all your nodes may improve performance by virtually locking CPU cores to VINS-Mono. It is a theory...

shakith, 2018-05-22:
@maciejmatuszak, Thanks for the info.

tiralonghipol, 2018-08-30:
Hi. I recently bought the Optor cam, which I think is running the same sdk as Loitor, can you confirm this?

mzahana, 2018-10-04:
Hi @tiralonghipol

I just came across Optor cam, and I'd like to get some feedback about it's performance in a VIO system. Is there synchronization between the IMU and cameras?
Are they accurately timestamped?

Any issues you faced with it so far?

Thanks.

tiralonghipol, 2018-10-04:
Hi. 
After many tries I assumed that this device is not hardware synchronized as declared
here you can read more in deep what the problem is: https://github.com/ethz-asl/rovio/issues/192