HKUST-Aerial-Robotics/VINS-Mono #204: getting ros warn imu message in disorder

Razbotics, 2018-05-19:
i am continuously getting warning message saying  imu message in disorder...why is it so...also the vins estimation drifts away very fast . Is it due to incorrect imu bias?

jn-c, 2018-07-09:
Have you fixed that yet? I got the same problem.

srikar8, 2018-08-09:
@Razbotics have you solved this?
thanks in advance.

Razbotics, 2018-08-09:
This is due to incorrect timestamps and large unsynchronisation between imu and image frames

chenwinki, 2018-11-15:
@Razbotics  Hello, how did you deal with the problem of "imu message in disorder!",  could you give me a hint? Thank you!

srikar8, 2018-11-15:
@chenwinki check the time difference of imu and image timestamp and assign td in the config file.

chenwinki, 2018-11-15:
@srikar8 Thank you ! But I assign the td in the config file by calibrate imu and camera using kalibr, but I still get the warning "imu message in disorder", I also tried give a  initial td and online estimate time offset, it does not work either.

innovatim, 2018-11-15:
@chenwinki  Please try checking the code that sets the timestamp in the imu driver.

chenwinki, 2018-11-15:
@innovatim Thank you! I will try your advice, actually, i use ZED MINI, and the driver zed_wapper, I will spend some time to read the code.

JUGGHM, 2018-12-24:
@chenwinki Hello! I'd like to ask whether you ran VINS successfully using ZED-mini? I'm considering buying a ZED-mini as a VI sensor.

mildwinter, 2019-11-12:
You can try to adjust the frequency of the imu release. I solved this warning by this method, but recently I found that the odometer drift is more serious. I haven't determined whether it is caused by this change.

varun3108, 2020-11-03:
I am getting the same warning for playing the rosbag I had created from https://github.com/OSUPCVLab/mobile-ar-sensor-logger/wiki

zhaoworking, 2020-11-20:
> You can try to adjust the frequency of the imu. I solved this warning by this method, but recently I found that the odometer drift is more serious. I haven't determined whether it is caused by this change.

I just want to know how you reduce your imu stream frequency , is that just reducing the imu data timestamps in file `imu0.csv`?