HKUST-Aerial-Robotics/VINS-Mono #206: About the came intrinsic prameter

dpfzju, 2018-05-29:
I have downloaded the data of Euroc in zip format, and there is several yaml files. But I found that the camera parameters is different with parameter in euroc_config.yaml. Here is the camera intrinsic parameter in sensor.yaml about cam0: 
    intrinsics: [458.654, 457.296, 367.215, 248.375] #fu, fv, cu, cv
    distortion_model: radial-tangential
    distortion_coefficients: [-0.28340811, 0.07395907, 0.00019359, 1.76187114e-05]
  But in euroc_config.yaml,  they are: 
    distortion_parameters:
       k1: -2.917e-01
       k2: 8.228e-02
       p1: 5.333e-05
       p2: -1.578e-04
   projection_parameters:
       fx: 4.616e+02
       fy: 4.603e+02
       cx: 3.630e+02
       cy: 2.481e+02
    Why are they different?

yeukfu, 2018-08-23:
I also found the difference. What is strange is that with parameters in euroc_config.yaml I got more precise estimation output.

AIgithub22, 2018-10-02:
the parameters in euroc_config.yaml is calibrated by Kalibr witch is the calibration tools vinsmono recommended.

XiaoxingChen, 2019-09-26:
> the parameters in euroc_config.yaml is calibrated by Kalibr witch is the calibration tools vinsmono recommended.
EuroC and Kalibr are both released by ETHZ-ASL, I don't think the senser.yaml from EuroC was calibrated by tools other than Kalibr.