HKUST-Aerial-Robotics/VINS-Mono #215: Using true depth with VINS-Mono

yukti-nxg, 2018-07-02:


mzahana, 2018-08-28:
@yukti-nxg Hi. I have D435 as well, but I am not uste how to sync the depth and IMU and use it with VINS MONO. Can you please give some hints? Thanks.

RigerLee, 2018-09-08:
I recommend message_filters for sync.
Substitute
`ros::Subscriber sub_img = n.subscribe(IMAGE_TOPIC, 100, img_callback);` 
in feature_tracker_node by 
`message_filters::Subscriber<sensor_msgs::Image>`` color_sub(n, IMAGE_TOPIC, 1);`
`message_filters::Subscriber<sensor_msgs::Image> depth_sub(n, DEPTH_TOPIC, 1);`
`message_filters::TimeSynchronizer<sensor_msgs::Image, sensor_msgs::Image> sync(color_sub, depth_sub, 100);`
`sync.registerCallback(boost::bind(&img_callback, _1, _2));`
(I modified img_callback for depth transmission.)
And then you could do futher modification.

jackyfriend, 2019-10-22:
@yukti-nxg Hi, I use Realsense D435 too, I'm wandering how to add a synced IMU with d435's color image, could you give me some advices? Thank you!