HKUST-Aerial-Robotics/VINS-Mono #218: Encountered an error when I used camera

jn-c, 2018-07-12:
Hi everyone. A couple days ago, I received an imu and a camera from my tutor, then I followed the [instruction](https://github.com/HKUST-Aerial-Robotics/VINS-Mono#5-run-with-your-device) to modify the topic names in the config file. After that, I started to run VINS-Mono, but the terminal had such an output:

> OpenCV Error: Assertion failed (_mask.empty() || (_mask.type() == (((0) & ((1 << 3) - 1)) + (((1)-1) << 3)) && _mask.sameSize(_image))) in goodFeaturesToTrack, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgproc/src/featureselect.cpp, line 366
> terminate called after throwing an instance of 'cv::Exception'
>   what():  /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/imgproc/src/featureselect.cpp:366: error: (-215) _mask.empty() || (_mask.type() == (((0) & ((1 << 3) - 1)) + (((1)-1) << 3)) && _mask.sameSize(_image)) in function goodFeaturesToTrack
> 
> [feature_tracker-2] process has died [pid 13709, exit code -6, cmd /home/dauker/catkin_ws/devel/lib/feature_tracker/feature_tracker __name:=feature_tracker __log:=/home/dauker/.ros/log/34b0daf2-85d8-11e8-a2ad-d05349d44da1/feature_tracker-2.log].
> log file: /home/dauker/.ros/log/34b0daf2-85d8-11e8-a2ad-d05349d44da1/feature_tracker-2*.log

  After I executed command roslaunch vins_estimator euroc.launch, the above error occurred when I started the camera driver(the driver I used is [usb_cam](http://wiki.ros.org/usb_cam)). 
  Does anyone know how to solve this ? Any suggestions would be greatly appreciated.

thornsu, 2018-07-19:
do you set fish eye to 1 in config? set 0 and try again

jn-c, 2019-01-21:
@thornsu I am using the default value of 0.

closed 2019-01-22 by jn-c